UWRobotRacing/RobotRacing2018 #8: Create test tracks for the simulated robot

Oluwatoni, 2018-04-12:


Oluwatoni, 2018-05-13:
Drag race courtesy of @JXproject 
[image]

JXproject, 2018-05-19:
New Drag Race
[image]

JXproject, 2018-05-19:
Oval Race Track

[image]

JXproject, 2018-05-19:
Oval Race Track Detail shot at start line
[image]

JXproject, 2018-05-19:
Circuit Race track

[image]

JXproject, 2018-05-19:
Circuit Race Perspectve of the entrance of the shortcut

[image]

JXproject, 2018-05-19:
Circuit Race Perspective view of the starting line
[image]

closed 2018-05-19 by JXproject