UWRobotRacing/RobotRacing2018 #32: Raceday Checks!

Oluwatoni, 2018-06-24:
We need to put together a raceday checklist for the competition to allow us to verify that all the subsystems are ready to go.

Compile a list of checks/tests across the software nodes and the electrical team that we should do before each test run to prepare for the race.

angegu, 2018-06-28:
From Matt:
- LIPO batteries + charger and extra swap pair
- toolkit of (screwdrivers, all allen keys, phillips, have tools per fastener)
- zipties
- wire-cutters
- solder station
- run program before (day/ hour)
- computer can see simultaneously all peripheral devices (because usb hub)
- meets height and weight requirements (hour before!)
- run all software simultaneously
- check camera alignment
- * everything is labelled *
- make sure latest code is on git (everyone has copy)
- one extra laptop that can run all code
- centre the steering servos
- tightness of fasteners
- launch file checks (cascading)
- git branch is correct
- ppt is good 
- multiple copies of ppt
-

Oluwatoni, 2018-06-28:
Please make this into a seperate readme and put it in our repo home directory

angegu, 2018-07-03:
added new branch prerace_checks